CentraleNantesRobotics/ping360_sonar #7: New published topic

tomlogan501, 2019-12-20:
Hi,
It would be great to offer a new kind of topic in this node such as laserscan msg.

Regards,
Thomas

Stormiix, 2019-12-20:
@tomlogan501 Should be interesting! The initial purpose of the project was to do image processing on the output images. But we are open to add new topics and msg types.
If you want, you can make a pull request and we'll be happy to review it and add it to the project.

Regards,
Anas

tomlogan501, 2019-12-20:
I'm not a specialist of Python but I will try

Stormiix, 2019-12-20:
@tomlogan501 Don't worry about it. We'll help!

tomlogan501, 2019-12-20:
@Stormiix , by any chance, do you know the rotate frequency of the ping360 ? The datasheet is kind of mysterious about that . From 9s to 35s for one turn ?! I don't get it.

Stormiix, 2019-12-20:
> @Stormiix , by any chance, do you know the rotate frequency of the ping360 ? The datasheet is kind of mysterious about that . From 9s to 35s for one turn ?! I don't get it.

Yes, it depends on the range you've set (1m-50m) it takes from (9s-35s) per revolution.

tomlogan501, 2019-12-20:
Hmm, do you think it's linear ? or does it obey to some sort of water physic property ?

Stormiix, 2019-12-20:
@tomlogan501 Not sure, but looking at the way it's presented in the data sheet. It's safe to assume it's linear. But you should verify that experimentally.
By the wat, I've looked at the LaserScan msg type, it's similar to the sonar output, but not quite. The sonar msg would contain:
```
Header header            #header info

float32 angle               # the measurement angle [rad]
float32 scan_time        # scan duration [seconds]
float32 range      #  range value [m]
float32[] intensities    # intensity data [device-specific units].  This is the actual data received from the sonar
... the rest of the parameters, like: speed of sound, transmit frequency, gain ..etc
```

Stormiix, 2019-12-23:
Closed. Follow up on #11

closed 2019-12-23 by Stormiix

Stormiix, 2019-12-24:
@all-contributors please add @tomlogan501 for ideas