CentraleNantesRobotics/ping360_sonar #21: Unit tests and/or integration tests (with ROS)

tomlogan501, 2020-01-09:
**Is your feature request related to a problem? Please describe.**
No test, no coverage

**Describe the solution you'd like**
Add test directory and coverage support

**Additional context**
Use of python coverage module ? Does it generate a rapport ?

Stormiix, 2020-01-24:
Thanks for the bug reports, tests and feature suggestions. Let's add tomlogan501 for his contributions. cc @all-contributors

Stormiix, 2020-01-24:
@all-contributors please add @tomlogan501 for tests, bugs and features.

Stormiix, 2020-01-24:
Now that we have CI in place, tests would be an awesome thing to add. If you have any suggestions please do not hesitate!

tomlogan501, 2020-01-24:
We  could start with simple unit tests for common use verification and then add integration tests with ROS with emulated sensor and/or gazebo.

Stormiix, 2020-01-24:
Yes definitely. I will try to improve the emulator #33 once I'm done with the #34.

closed 2020-04-02 by github-actions[bot]

tomlogan501, 2020-04-02:
What ? It should be reopened