CentraleNantesRobotics/ping360_sonar #28: RViz LaserScan not publishing the Points

Hameck, 2020-01-15:
**Describe the bug**
When we launch RViz to see the LaserScan topic, we can note that there are messages being published in the topic /ping360_node/sonar/scan, but no Points appear, even when threshold is set to 0.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the files of the 'gaussian' branch
2. Run 'roslaunch ping360_sonar example.launch'
3. Run 'rqt' in another tab
4. Open Rviz
5. Select 'sonar_frame' as the fixed frame
6. Click in 'Add' to display a new information
7. Display the topic /ping360_node/sonar/scan as LaserScan
8. Open the dynamic reconfiguration windows, and set threshold to 0.
9. No Points can be seen.

**Expected behavior**
I was expecting to see a circular cloud of points.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - ROS: melodic
 - Python: python2.7
 - Version: [e.g. 1.0]

Stormiix, 2020-01-15:
While debugging and trying to fix this, I've multiple bugs caused by the global variables and the fact that parameters are fetched before the node initialization. 
Moreover, the dynamic reconfigure overwrites the parameters set in the launch file on load, making them useless.
Another bug was caused by the intensities/ranges list that weren't resized when the step param is modified.  

This gaussian branch should be merged asap.

closed 2020-01-15 by Stormiix

Hameck, 2020-01-15:
I saw this problem of the step, this was what I wanted to talk to you at Centrale

commit 12374481 (2020-01-15): Bugfixes - Closes #28

Major Bugfixes