CentraleNantesRobotics/ping360_sonar #33: Random noise for simulated signal

tomlogan501, 2020-01-20:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Add a parameter to generate either random noise (with another parameter fro amplitude) or square shape or both.

Stormiix, 2020-01-24:
Can you provide examples of the expected results (what would the image output look like?)?

Thanks.

tomlogan501, 2020-01-27:
An image would look like this.

[image]

instead of this 

[image]

It will be a complement to gazebo testing.

Hameck, 2020-01-29:
In the develop branch, I've added something to test (it's not completely random, but it is not a circle - which is useful for debugging). I would like to add something related to a perlin noise, but I couldn't find a lib to do so.

tomlogan501, 2020-01-30:
You have the pseudo code on wikipedia :) https://fr.wikipedia.org/wiki/Bruit_de_Perlin

Stormiix, 2020-01-30:
Hello @tomlogan501, I heard you tried to push something on the repo but you couldn't.
What you need to do is to push those changes to your forked repo and then create a new pull request, which we can then review and approve.
Also, don't forget to pull the latest changes :D

Thanks in advance! We really appreciate your help!

Hameck, 2020-01-30:
@tomlogan501 Checkout the branch `perlin_noise`. I've added the perlin noise in the generation of figures, having something very similar to the image you proposed.
[image]

closed 2020-01-30 by Hameck

tomlogan501, 2020-02-13:
Did you push your branch ?

Stormiix, 2020-02-14:
@tomlogan501  it's already merged into master, so you can already try it out using the release or master branch :D 
let us know if it's not working.

tomlogan501, 2020-02-17:
@Stormiix The emulated laserscan works