CentraleNantesRobotics/ping360_sonar #48: New node : from raw data to laser_scan (for replay and conversion)

tomlogan501, 2020-02-24:
**Is your feature request related to a problem? Please describe.**
Ability to convert data from a bag after an acquisition

**Describe the solution you'd like**
A simple node with parameters for the conversion raw <--> laser_scan or others

**Describe alternatives you've considered**
Modify the main node but to risky to change the code, better practice to create a new one

tomlogan501, 2020-02-24:
Code a thing that do the [trick](https://github.com/tomlogan501/ping360_sonar_python/tree/feature_48)

Stormiix, 2020-02-25:
That is doable, I'll make something once #49 is resolved.

Also I noticed in your code that implement an angleStart, we added something similar in our latest release (1.3.0). (Don't forget to pull the latest changes to your local repo)

We added two parameters, minAngle / maxAngle that allow you to set the start and end scan angles. Please note that these angles are in GRAD (I know it's confusing but we're trying to stay true to the BRping unit choices).

P.S: to always have the latest updates, make sure you're checked out on the develop branch. If you want the latest **stable** changes, use the _master_ or _release_ branches.

tomlogan501, 2020-02-26:
I'm well aware of the git workflow but the conversion you used isn't correct from rad to degree.
I push another node for the convert on my fork if you want to check out or use it as patch.

tomlogan501, 2020-02-26:
I create it for the threshold issue : I send bags to henrique about that 
You can check it with him

closed 2020-05-04 by github-actions[bot]