CentraleNantesRobotics/ping360_sonar #49: Remove previous angle scan

tomlogan501, 2020-02-24:
**Is your feature request related to a problem? Please describe.**
The topic scan is a merge of scans done at different time (Sonar rate is very slow)
Therefore the output topic is not reliable

**Describe the solution you'd like**
Just provide the current angle scan

**Describe alternatives you've considered**
Another possibility would be to create a new type of message like laserscanTimestamped but too much work.

Stormiix, 2020-02-25:
If we remove the previous scan angles, the node will only publish a single value ranges array containing the measured range at the current angle scan. 

Would that behavior be acceptable ? (I haven't used the LaserScan sensors before so I wonder if a single measure per message is the expected behavior)

tomlogan501, 2020-02-26:
I think this is acceptable, then you should use it like me with laser_assembler.

PR #55 (2020-03-27): Removing previous scans
Closes #49
closed 2020-03-27 by Stormix