CentraleNantesRobotics/ping360_sonar #52: Sector scan

tomlogan501, 2020-03-13:
**Describe the bug**
Sector scan doesn't correspond well to the ping viewer

**To Reproduce**
Steps to reproduce the behavior:
1. Create a launcher with maxAngle = 180
1. Launch it and rqt_gui
1. Observe that the sector scan doesn't equal 180° 

**Expected behavior**
Sector scan equal 180°

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - ROS: melodic
 - Version : 1.3.2 (head master)

**Additional context**
Tests done in swimming pool, only angle was modified from example.launch

Hameck, 2020-03-13:
Dear @tomlogan501 , in order to have a 180° scan, you should use a range of 200 gradians! you can always use the conversion : gradian = degrees*(10/9). Just a reminder : our program uses gradian and not degrees for the angles.

tomlogan501, 2020-03-13:
Ok I will test again, sorry for the trouble

tomlogan501, 2020-03-13:
Maybe you should abstract this conversion for the common user, even if you mention it.
The ROS Standard would be radian.
This issue can be change to new feature.

closed 2020-05-21 by github-actions[bot]