CentraleNantesRobotics/ping360_sonar #57: error from running "roslaunch ping360_sonar example.launch"

WYaomei, 2020-08-19:
When I run the ping360_sonar in my ROS melodic with python 2.7, I got the following error. I am wondering whether this is because of the version of python? thanks.
===================================================
Traceback (most recent call last):
  File "/home/ubuntu/ping360/src/ping360_sonar/nodes/ping360_node", line 3, in <module>
    from ping360_sonar import main
  File "/home/ubuntu/ping360/devel/lib/python2.7/dist-packages/ping360_sonar/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
ImportError: cannot import name 'main'
====================================================

Stormix, 2020-08-19:
It's most likely that, did you try running it on python 3 ?

WYaomei, 2020-08-19:
I am trying to build an virtual environment with python 3 on ros melodic now.  Will try to run on python 3 which the environment is built.

WYaomei, 2020-08-20:
When I use python 3, another error shows 
=============================
Traceback (most recent call last):
  File "/home/ubuntu/ping360/src/ping360_sonar/nodes/ping360_node", line 3, in <module>
    from ping360_sonar import main
ModuleNotFoundError: No module named 'ping360_sonar'
==============================

could you please tell me how you run the code in ROS, any setting? which version of ROS you use? I think the default version of python for Melodic is python 2. How did you change to python 3? Thanks.

VV-AM, 2020-09-14:
Hi There,

I got __ImportError: No module named brping__ while launching :(
Any idea for the reason? 
I am working via virtualbox - could this be a reason?
I had to install bluerobotics package with user option due to acces to serial erroe, but anyway it is installed already.

Kind regards

VV

Stormix, 2020-09-14:
@VV-AM I would appreciate it if you create a separate issue for this. But I'll answer on this thread anyways. Could you verify on what python version the package is installed ? And what version your ROS setup is using ?

VV-AM, 2020-09-14:
As adviced I've created new issue for this #59

closed 2022-04-07 by github-actions[bot]