CentraleNantesRobotics/ping360_sonar #58: New version of blue robotics package

tomlogan501, 2020-09-14:
**Describe the bug**
Due to a new version of blue robotics package, the example launch craches

**To Reproduce**
Steps to reproduce the behavior:
1. Update to package blue robotics 0.1.0
2. Launch example.launch

**Expected behavior**
Normal beahavior

**Screenshots**
```
SUMMARY
========

PARAMETERS
 * /ping360_node/baudrate: 115200
 * /ping360_node/debug: False
 * /ping360_node/device: /dev/ttyUSB0
 * /ping360_node/enableDataTopic: True
 * /ping360_node/enableImageTopic: True
 * /ping360_node/enableScanTopic: True
 * /ping360_node/gain: 0
 * /ping360_node/imgSize: 500
 * /ping360_node/maxAngle: 400
 * /ping360_node/minAngle: 0
 * /ping360_node/numberOfSamples: 200
 * /ping360_node/oscillate: True
 * /ping360_node/queueSize: 1
 * /ping360_node/sonarRange: 10
 * /ping360_node/speedOfSound: 1500
 * /ping360_node/step: 1
 * /ping360_node/threshold: 100
 * /ping360_node/transmitFrequency: 740
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    ping360_node (ping360_sonar/ping360_node)
    transform_sonar_frame (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [22236]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 81e973f6-f66a-11ea-8c6f-23186ea11dc8
process[rosout-1]: started with pid [22247]
started core service [/rosout]
process[transform_sonar_frame-2]: started with pid [22250]
process[ping360_node-3]: started with pid [22255]
Traceback (most recent call last):
  File "/home/tsimon/ping_ws/src/ping360_sonar/nodes/ping360_node", line 3, in <module>
    from ping360_sonar import main
  File "/home/tsimon/ping_ws/devel/lib/python2.7/dist-packages/ping360_sonar/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/home/tsimon/ping_ws/src/ping360_sonar/src/ping360_sonar/node.py", line 15, in <module>
    from sensor import Ping360
  File "/home/tsimon/ping_ws/src/ping360_sonar/src/ping360_sonar/sensor.py", line 6, in <module>
    from brping import definitions
  File "/home/tsimon/.local/lib/python2.7/site-packages/brping/__init__.py", line 4, in <module>
    from brping.device import PingDevice
  File "/home/tsimon/.local/lib/python2.7/site-packages/brping/device.py", line 38
    def connect_serial(self, device_name: str, baudrate: int =115200):
                                        ^
SyntaxError: invalid syntax
[ping360_node-3] process has died [pid 22255, exit code 1, cmd /home/tsimon/ping_ws/src/ping360_sonar/nodes/ping360_node __name:=ping360_node __log:=/home/tsimon/.ros/log/81e973f6-f66a-11ea-8c6f-23186ea11dc8/ping360_node-3.log].
log file: /home/tsimon/.ros/log/81e973f6-f66a-11ea-8c6f-23186ea11dc8/ping360_node-3*.log

```

**Desktop (please complete the following information):**
 - ROS: [melodic]
 - Python [python2.7]
 - Bluerobotics : 0.1.0
 - Version [e.g. master]

Should add on the ReadMe what version of bluerobotics should be used.

Stormix, 2020-09-14:
They finally updated the BR python package. I'll try to look into it this week. Might take a while before I make an update.
We might also need to update the package to use Python3 and resolve #57 .

tomlogan501, 2020-09-14:
Do they plan to do other versions or is it the final ?
I come back to 0.0.9 for my swimming pool tests. 
What could we add in the workflow to prevent this ? Integration tests with an emulator ?

Stormix, 2020-09-14:
Their update cycle is quite random, ideally, I should fetch the latest definitions from [here](https://github.com/bluerobotics/ping-python/blob/deployment/brping/device.py) and programmatically add our code changes. Or like you have proposed, specify the package version on the readme.

Integration tests would definitely help us detect these kind of breaking changes. I should probably add some later on. Meanwhile, since 0.0.9 worked for you, I'll temporarily add it to the readme.

Thanks again for the quick feedback @tomlogan501 , hope you're doing fine during this crisis.

tomlogan501, 2020-09-14:
No problem, that's my contribution :) 
Hope you're doing well too

closed 2020-09-14 by tomlogan501