CentraleNantesRobotics/ping360_sonar #59: No module brping while launching example.launch

VV-AM, 2020-09-14:
Hi There,

I got _ImportError: No module named brping_ while launching :(
Any idea for the reason?
I am working via virtualbox - could this be a reason?
I had to install bluerobotics package with user option due to acces to serial erroe, but anyway it is installed already.

Python version: 2.7.12
ROS version: kinetic

Kind regards

VV

Stormix, 2020-09-15:
@VV-AM  Could you verify on what python version the package is installed ? And what version your ROS setup is using ?

VV-AM, 2020-09-15:
I put above info: 
Python version: 2.7.12
ROS version: kinetic

Do we need any more details (I don't know how to get them in fact :( )

rgds

VV

jae-hyeong-hwang, 2021-04-11:
How did this issue go ?
for me, my default python version is python3.6
also have python2.7 but it was necessary for ros-melodic
for me, I have this error

> Traceback (most recent call last):
  File "/home/jaehh_re/catkin_ws/src/ping360_sonar/nodes/ping360_node", line 3, in <module>
    from ping360_sonar import main
  File "/home/jaehh_re/catkin_ws/devel/lib/python2.7/dist-packages/ping360_sonar/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
ImportError: cannot import name 'main'
> 
```diff
- [ping360_node-2] process has died [pid 5818, exit code 1, cmd /home/jaehh_re/catkin_ws/src/ping360_sonar
- /nodes/ping360_node __name:=ping360_node __log:=/home/jaehh_re/.ros/log/abc70136-9ac3-11eb-a18d-f8a2d6832c79
- /ping360_node-2.log].
- log file: /home/jaehh_re/.ros/log/abc70136-9ac3-11eb-a18d-f8a2d6832c79/ping360_node-2*.log

tomlogan501, 2021-05-20:
seems to be the same issue here #64

closed 2022-04-14 by github-actions[bot]