CentraleNantesRobotics/ping360_sonar #60: Invalid syntax - process has died (roslaunch launch file)

kamkow8, 2020-12-22:
I get this message when running example.launch.

Traceback (most recent call last):
  File "/home/kamil/Workspaces/sonar_ws/src/ping360_sonar/nodes/ping360_node", line 3, in <module>
    from ping360_sonar import main
  File "/home/kamil/Workspaces/sonar_ws/devel/lib/python2.7/dist-packages/ping360_sonar/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/home/kamil/Workspaces/sonar_ws/src/ping360_sonar/src/ping360_sonar/node.py", line 15, in <module>
    from sensor import Ping360
  File "/home/kamil/Workspaces/sonar_ws/src/ping360_sonar/src/ping360_sonar/sensor.py", line 6, in <module>
    from brping import definitions
  File "/home/kamil/.local/lib/python2.7/site-packages/brping/__init__.py", line 4, in <module>
    from brping.device import PingDevice
  File "/home/kamil/.local/lib/python2.7/site-packages/brping/device.py", line 38
    def connect_serial(self, device_name: str, baudrate: int =115200):
                                        ^
SyntaxError: invalid syntax
**[ping360_node-2] process has died [pid 5545, exit code 1, cmd /home/kamil/Workspaces/sonar_ws/src/ping360_sonar/nodes/ping360_node __name:=ping360_node __log:=/home/kamil/.ros/log/b15eefc2-4443-11eb-a5e9-0800279a33a9/ping360_node-2.log].
log file: /home/kamil/.ros/log/b15eefc2-4443-11eb-a5e9-0800279a33a9/ping360_node-2*.log**

What I did:
- pip install bluerobotics-ping
- I created the folders: sonar_ws> src
- cd sonar_ws / src
- git clone https://github.com/CentraleNantesRobotics/ping360_sonar.git
- cd ../
- catkin_make
- in launch file: <env name = "emulated_sonar" value = "true" />
- roslaunch ping360_sonar example.launch

I only have such files in the given path:
master.log                   roslaunch-kamil-vb-5449.log  rosout-1-stdout.log
roslaunch-kamil-vb-4884.log  roslaunch-kamil-vb-5491.log  rosout.log
roslaunch-kamil-vb-5116.log  roslaunch-kamil-vb-5529.log  transform_sonar_frame-1-stdout.log

Stormix, 2020-12-22:
It might be that your sonar is not plugged in to your computer, or is on a different port, try changing the `/dev/ttyUSB0` to the right USB port.

Stormix, 2020-12-22:
Please reopen if you feel like it's not the case.

closed 2020-12-22 by Stormix

kamkow8, 2020-12-25:
Yes, the sonar is not connected, but I wanted to run an emulator.
However, despite changing false to true in:
<env name = "emulated_sonar" value = "true" />
the emulator does not start. Can it be turned on without connecting a sonar?