CentraleNantesRobotics/ping360_sonar #66: Could you give the some detailed information on the topic that /ping360_node/sonar/scan?

tank179, 2021-08-29:
roslaunch ping360_sonar example.launch can run correctly, and the topic that /ping360_node/sonar/scan can be published and received. But I do not understand the the physical significance of the message that belongs to message type that sensor_msgs/LaserScan from the topic that /ping360_node/sonar/scan: 
float32 angle_min = minAngle          # start angle of the scan [rad]
float32 angle_max = maxAngle         # end angle of the scan [rad]
float32 angle_increment   # angular distance between measurements, calculated using the configured Step  [rad]
float32 range_min = .75                     # minimum range value [m]
float32 range_max = sonarRange      # maximum range value, it's set to the configured sonarRange [m]
float32[] ranges                                  # calculated ranges that correspond to an intensity > threshold [m] 
float32[] intensities                             # sensor intensity data [0-255].
especially for these elements that float32 angle_increment, float32[] ranges and float32[] intensities, and the relations among these elements.
Thank you very much!

Stormix, 2021-09-02:
Try reading through this project report and let me know if you still have questions.

[report.pdf](https://github.com/CentraleNantesRobotics/ping360_sonar/files/7098863/report.pdf)

Stormix, 2021-09-05:
Closing for now, let me know if you have any other questions.

closed 2021-09-05 by Stormix

tank179, 2021-09-07:
Thank you for your reply! I will try reading through this project report. I will trouble you, if I still have questions. Thanks a lot!

tank179, 2021-09-11:
I have read the project report and understand it in general. I have another question to ask you that could you give me some explanation or a example that how to get distance and orientation of the obstacle ( assuming that only one obstacle in front of the ping360 sonar) according to /ping360_node/sonar/data or /ping360_node/sonar/scan. 
Thanks a lot.