CentraleNantesRobotics/ping360_sonar #67: ROS2 version

Rezenders, 2022-03-22:
**Describe the bug**
I am trying to build ping360_sonar package for ROS2 but it is not working.

I get the following error:

```
Starting >>> ping360_sonar_msgs
Finished <<< ping360_sonar_msgs [0.55s]                     
Starting >>> ping360_sonar
--- stderr: ping360_sonar                         
CMake Error at CMakeLists.txt:23 (add_subdirectory):
  The source directory

    /home/gus/bluerov_ros2_ws/src/ping360_sonar/ping360_sonar/src/ping-cpp/src

  does not contain a CMakeLists.txt file.

CMake Error at /opt/ros/foxy/share/ament_cmake_core/cmake/symlink_install/ament_cmake_symlink_install_targets.cmake:67 (message):
  ament_cmake_symlink_install_targets() 'ping-cpp' is not a target
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_core/cmake/symlink_install/install.cmake:37 (ament_cmake_symlink_install_targets)
  CMakeLists.txt:40 (install)

---
Failed   <<< ping360_sonar [0.71s, exited with code 1]

Summary: 1 package finished [1.36s]
  1 package failed: ping360_sonar
  1 package had stderr output: ping360_sonar

```

**To Reproduce**
```
cd ros2_workspace/src
git clone https://github.com/CentraleNantesRobotics/ping360_sonar.git -b ros2 --recursive
cd ..
colcon build --symlink-install
```

**Desktop (please complete the following information):**
 - ROS: foxy
 - Python: 3.8.10
 - Version: ros2 branch

**Additional questions**
Is the ros2 version working?
Do you guys recommend using it or should I stick with the ros1 version?

oKermorgant, 2022-03-22:
Hi, 

Thanks for your interest. The ROS 2 branch is under heavy development and is not working yet.
We hope the Python version will ship in a few days - it is part of a student project here.

The compilation error was coming from the C++ version, when adapting to the latest [ping-cpp](https://github.com/GSO-soslab/ping-cpp/tree/14183a9d6cd484f7009f174fc88c8efa1c458c8a) libs.
I have just pushed a commit that makes it compile, though the C++ version still does nothing yet. 

Our ultimate goal is to favor C++ in order to save CPU / battery, as embedded Python can quickly submerge our Raspberry.

Rezenders, 2022-03-22:
Thanks for the reply!

I understand. Anyway, nice to know that you guys are actively working on it. I will keep checking it, and for now I will stick with the ROS1 version. Maybe I can help to test it if necessary.

closed 2022-04-06 by Stormix