CentraleNantesRobotics/ping360_sonar #76: trigger sonar slow

linzha0, 2022-04-14:
Hi, thanks for sharing this awesome project. Currently, We modified your code from ROS2 to ROS1 for our vehicle. Basically, just replace the ROS2 wrap, the interface code is same and it works well. The code is [here](https://github.com/GSO-soslab/bluerobotics_ping360)

But we found a issue that sonar trigger is kind of slow. For basic setup, it takes nearly 48 second to rotate 360 degree, the ping-viewer takes 6 seconds in same configuration. we tested the sonar in a trash can within water. 

we found `sonar.waitMessage(Ping360Id::DEVICE_DATA, 8000)` from `sonar_interface/Ping360Interface::read()` will take about 0.11 second. maybe that's the problem. when we reduce timeout less then 300, it become faster but lost some data.

We just wondering if you have conditions like this before. If you could give us some suggestions, it will be great. thanks for the help.

AlexisFetet, 2022-04-18:
Hi,

Thanks for your interest.
We are currently using [this](https://github.com/GSO-soslab/ping-cpp/blob/14183a9d6cd484f7009f174fc88c8efa1c458c8a/src/device/ping-device.cpp) definition of the function, I am usure if this is the reason for the performance gap as this part is ROS independant. 
I noticed that you are using the same timeout value for your ROS version (`sonar.waitMessage(Ping360Id::DEVICE_DATA, 8000)`), did this still improve performance?

I am personnally not able to perform testing on an actual sonar, however I'm interested in knowing what part is actually taking this much time. Did you try timing each major function as you did with the `waitMessage`? This would help knowing where to focus our attention in the first place.

AlexisFetet, 2022-04-18:
Regarding the timeout, I created a request to add it as parameter for more customizability. #77

oKermorgant, 2022-04-29:
Hi, 

The PR was just merged, we still need to do tests with the sonar.
Up to now we have mainly focused on the configuration / communication / display.

AlexisFetet, 2022-05-31:
Hi, 
@linzha0, @Rezenders, would it be possible to have more information about the hardware you tested the node on? And possibly provide the CPU and RAM load during the tests, as well as other running programs during said tests?

ES-Alexander, 2022-06-17:
@Rezenders (/ @AlexisFetet )

> I figured out that the reason for this is that there is a bridge between `/dev/ttyUSB0` and `0.0.0.0:9092`.

Assuming you're using [BlueOS](https://github.com/bluerobotics/BlueOS-docker) on your Raspberry Pi, that will be from [the `ping` service](https://github.com/bluerobotics/BlueOS-docker/tree/master/core/services/ping) which is made to look for serial Ping devices and set up bridges to allow them to be accessed via UDP (including from the topside).

Since you've managed to get a UDP connection working with ROS, did that help with your speeds?

---

More generally on the speed front, the scan speed depends on the scan parameters (e.g. longer distances and finer angular resolution take longer), as well as the speed of the I/O to send and receive messages (at both the physical and software levels - ethernet is faster than USB, and C++ is faster than Python).

To find out what speeds you can expect it may be worth connecting to the Ping360 via UDP from the topside using [ping-python](https://github.com/bluerobotics/ping-python). There's [some example code in ping360.py](https://github.com/bluerobotics/ping-python/blob/deployment/brping/ping360.py#L279-L311) that can measure how fast you can expect to receive data with a given set of settings. If you use the same settings there as for your ROS setup you can determine whether it's a general sonar or Python issue, or if there are slowdowns from the ROS integration.

closed 2022-08-08 by AlexisFetet