CentraleNantesRobotics/ping360_sonar #78: Is it possible to connect to the sonar via udp?

Rezenders, 2022-05-12:
Hi,

Is it possible to connect to the sonar via udp? instead of the serial port?
I would like to run this package on a computer connected to the bluerov.

I was trying to do it like this:
`ros2 run ping360_sonar ping360_node --ros-args --param device:=udp://192.168.2.2:9092`

AlexisFetet, 2022-05-20:
Hi, 

We only use the Serial connection so far, however there is a [UDP](https://github.com/GSO-soslab/ping-cpp/blob/14183a9d6cd484f7009f174fc88c8efa1c458c8a/src/hal/link/desktop/udp-link.cpp) link class as well.
An option would be to use another generator for the [sonar_interface](https://github.com/CentraleNantesRobotics/ping360_sonar/blob/ros2/ping360_sonar/src/sonar_interface.cpp), using maybe another set of arguments for UDP, having a generator for Serial and another for UDP.
I think a way to do it would be to have a `connexion` parameter set as `Serial` by default to match other default parameters.

AlexisFetet, 2022-05-21:
Hi, 
I believe the links are destroyed at the end of the constructor because you exit the scope they have been created in, resulting in not being able to use them later on. I tried the same method, with the following changes over yours :
 - replace `SerialLink serial_link` by `std::unique_ptr<SerialLink> serial_link;`
 - add `std::unique_ptr<UdpLink> udp_link;` 
 - use this conditionnal in constructor : 
```
if(connection_type.compare("serial") == 0)
{
      serial_link = std::make_unique<SerialLink>(device, baudrate);
      sonar = std::make_unique<Ping360>(*serial_link.get());
}
else if(connection_type.compare("udp") ==0)
{
      udp_link = std::make_unique<UdpLink>(device, std::to_string(baudrate));
      sonar = std::make_unique<Ping360>(*udp_link.get());
 }
```
Please let me know if this fixes your issue, and also if both serial and udp connexion are working as they do when manually replacing the connexion type (if you are able to test both).
If this solves the issue maybe we should introduce the `udp_port` parameter for the node, and give an extra parameter to the interface to stay consistant with the value names and roles.

Rezenders, 2022-05-23:
I just tested it for both the serial and udp connection and it worked. 
I added a `udp_address` and `udp_port` parameter as you suggested.
I opened a PR #81 with the changes. Let me know if you think I should modify anything.
Thanks for the help!

AlexisFetet, 2022-05-24:
I will close this issue as both python and C++ nodes received enhancement.