CentraleNantesRobotics/ping360_sonar #84: class SonarInterface has no attribute max_range

rajmis, 2023-04-19:
The ros2 node crashes on calling the `rangeFrom()` function after enabling the `/scan` topic. This happens because `SonarInterface` class has no attribute `max_range` -
https://github.com/CentraleNantesRobotics/ping360_sonar/blob/8fcd65db9a1de25d690c869d0e7bd51dace19c25/ping360_sonar/ping360_sonar/sonar_interface.py#L86

I am guessing that `configureTransducer()` function is missing the following declaration:
```python
self.max_range = max_range
```

AlexisFetet, 2023-04-21:
Hi, 
Thank you for pointing this out!
Could you please confirm that the following [fork](https://github.com/AlexisFetet/ping360_sonar) solves the issue for an actual sonar ? I am only able to test on emulated sonar.
Best regards,
Alexis

rajmis, 2023-04-24:
Yep, this has resolved the problem. Thanks!
Just a suggestion in case you don't merge with this repo, do update the [GitHub repo link](https://github.com/AlexisFetet/ping360_sonar/blob/c4c959e1b1b26c6266a87e7ff142689136e27da7/README.md?plain=1#L40) in the `README.md` of your forked version.

AlexisFetet, 2023-04-24:
Hi,
Perfect then. I will merge the fork then.

closed 2023-04-24 by AlexisFetet