CentraleNantesRobotics/ping360_sonar #86: Ros noetic

codeerzz, 2023-08-30:
does your package work in ros noeticte or what do we need to do to make it work?

AlexisFetet, 2023-09-12:
Hello,
As far as I know we do not have support for ROS Noetic. We have on the [master](https://github.com/CentraleNantesRobotics/ping360_sonar/tree/master) branch written some time ago for Melodic, you could follow this [tutorial](http://wiki.ros.org/noetic/Migration) to migrate it to Noetic. 
Alternatively, you could work on a backport of the current ROS2 version for Noetic.
I unfortunately do not have time to make a port myself at the moment.
Best regards,
AF.

closed 2023-09-18 by codeerzz

GabrielWellander, 2024-11-28:
Hello codeerz,
How did you solve your problem?

codeerzz, 2024-12-01:
We recently migrated the code from Python 2 to Python 3, and it works perfectly. The package is not currently on my computer, but I'll get it from my friends and upload a working version in a few days.

GabrielWellander, 2024-12-12:
Sorry for a late reply,
Where can I find that repo?
Or are you planning to make a pull request?