CentraleNantesRobotics/ping360_sonar #88: ERROR:colcon.colcon_cmake.task.cmake.build

phamquandung, 2023-09-26:
Hello authors, 

When I tried to build the code, I encountered this issue

**Describe the bug**
It said that: 
Failed to find the following files:
-/home/pqdung/ros2_workspace/install/ping360_sonar_msgs/share/ping360_sonar_msgs/package.sh
**To Reproduce**
Steps to reproduce the behavior:
1. I run: colcon build --symlink-install --packages-select ping360_sonar in the folder ~/ros2_workspace
2. See error
```
colcon build --symlink-install --packages-select ping360_sonar
Starting >>> ping360_sonar
[0.941s] ERROR:colcon.colcon_cmake.task.cmake.build:Failed to find the following files:
- /home/pqdung/ros2_workspace/install/ping360_sonar_msgs/share/ping360_sonar_msgs/package.sh
Check that the following packages have been built:
- ping360_sonar_msgs
Failed   <<< ping360_sonar [0.06s, exited with code 1]

Summary: 0 packages finished [0.53s]
  1 package failed: ping360_sonar
```

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - ROS: humble
 - Python python 3.10.12
 - Version 3

AlexisFetet, 2023-10-19:
Hi,

I see you ran `colcon build --symlink-install --packages-select ping360_sonar` which does not build the message package. did you build the message package beforehand? (`colcon build --symlink-install --packages-select ping360_sonar_msgs`)

AlexisFetet, 2023-10-20:
Hi,

`colcon build --symlink-install --packages-select ping360_sonar` only targets the `ping360_sonar` package and not its dependancies. please run `colcon build --symlink-install --packages-up-to ping360_sonar` to build `ping360_sonar` and its dependancies.
[image]
I am closing this issue as I believe this to be solved.

closed 2023-10-20 by AlexisFetet